ARMmbed/mbed-client-quickstart #21: What is this ?

ocZio, 2015-09-23:
```cpp
uint16_t port = rand() % 65535 + 12345;
```
source: https://github.com/ARMmbed/mbed-client-examples/blob/master/source/main.cpp#L85

yogpan01, 2015-09-23:
This is a listening port for incoming data but there is currently issue in DTLS handling on mbed Device Server side that it caches the port and will not allow re-connection requests on same port. So, we need to add randomization so that reseting and re-connecting this example will work.
Once issue is resolved on mbed Device Server end , we will no longer require to do this randomization of port value.

ocZio, 2015-09-23:
Thanks for the explanation yogplan01.

So what would happen if the port get's a larger value than 65535 ?

yogpan01, 2015-09-23:
It won't go above 65535 as the data type is uint16_t , it will wrap around :)

closed 2015-09-23 by ocZio