ARMmbed/mbed-client-quickstart #32: Baud rate on official examples

lws-team, 2015-11-01:
Hi -

This is a small thing... but the official examples should all use the same serial port baud rate.

mbed-example-network has this

    static Serial pc(USBTX, USBRX);
    pc.baud(115200);

but mbed-client-examples does not, so on the one officially supported board K64F, it defaults to 9600.  That means everyone switching between these examples, will think one or the other is not showing any output.

I can send a patch adding the above to mbed-client-examples if you like.

-Andy

mlnx, 2015-11-01:
+1
cc @bogdanm @meriac 
Can we make sure that all examples use the same baud rate? (115200)
Thanks

yogpan01, 2015-11-02:
This is now fixed and released. Example is using baud rate of 115200.

closed 2015-11-02 by yogpan01