ARMmbed/mbed-client-quickstart #33: Too many warnings at build time

BlackstoneEngineering, 2015-11-12:
## Problem
There are a bunch of unhandled warnings that need to be resolved on this repo

## Example 
Here is a screen cap of what I got from building this example for the `frdm-k64f-gcc` target

```shell
(workspace) C:\cygwin\home\ausbla01\mbed\mbed-client-examples>yotta --version
0.9.2

(workspace) C:\cygwin\home\ausbla01\mbed\mbed-client-examples>yt target frdm-k64f-gcc

(workspace) C:\cygwin\home\ausbla01\mbed\mbed-client-examples>yt build
info: get versions for sal-stack-lwip
info: download sal-stack-lwip@1.0.3 from the public module registry
info: get versions for sal-driver-lwip-k64f-eth
info: download sal-driver-lwip-k64f-eth@1.0.2 from the public module registry
info: get versions for sal-iface-eth
info: download sal-iface-eth@1.0.1 from the public module registry
info: generate for target: frdm-k64f-gcc 1.0.1 at C:\cygwin\home\ausbla01\mbed\mbed-client-examples\yotta_targets\frdm-k64f-gcc
GCC version is: 4.9.3
-- The ASM compiler identification is GNU
-- Found assembler: C:/yotta/gcc/bin/arm-none-eabi-gcc.exe
-- Configuring done
-- Generating done
-- Build files have been written to: C:/cygwin/home/ausbla01/mbed/mbed-client-examples/build/frdm-k64f-gcc
[13/319] Building C object ym/nanostack-libservice/source/CMakeFiles/nan...10f180/yotta_modules/nanostack-libservice/source/libip6string/stoip6.c.o
C:/cygwin/home/ausbla01/mbed/mbed-client-examples/yotta_modules/nanostack-libservice/source/libip6string/stoip6.c: In function 'stoip6':
C:/cygwin/home/ausbla01/mbed/mbed-client-examples/yotta_modules/nanostack-libservice/source/libip6string/stoip6.c:44:42: warning: comparison between
signed and unsigned integer expressions [-Wsign-compare]
     for (field_no = 0, p = ip6addr; (len > (p - ip6addr)) && *p && field_no < 8; p = q + 1) {
                                          ^
C:/cygwin/home/ausbla01/mbed/mbed-client-examples/yotta_modules/nanostack-libservice/source/libip6string/stoip6.c:54:18: warning: comparison between
signed and unsigned integer expressions [-Wsign-compare]
         if ((len > (q - ip6addr)) && *q && (q[0] == ':') && (q[1] == ':')) {
                  ^
[23/319] Building C object ym/mbed-client-c/source/CMakeFiles/mbed-clien...lient-examples/yotta_modules/mbed-client-c/source/libNsdl/src/sn_grs.c.o
C:/cygwin/home/ausbla01/mbed/mbed-client-examples/yotta_modules/mbed-client-c/source/libNsdl/src/sn_grs.c: In function 'sn_grs_list_resource':
C:/cygwin/home/ausbla01/mbed/mbed-client-examples/yotta_modules/mbed-client-c/source/libNsdl/src/sn_grs.c:145:84: warning: unused parameter 'pathlen'
 [-Wunused-parameter]
 extern sn_grs_resource_list_s *sn_grs_list_resource(struct grs_s *handle, uint16_t pathlen, uint8_t *path)
                                                                                    ^
[25/319] Building C object ym/mbed-client-c/source/CMakeFiles/mbed-clien...ples/yotta_modules/mbed-client-c/source/libCoap/src/sn_coap_protocol.c.o
C:/cygwin/home/ausbla01/mbed/mbed-client-examples/yotta_modules/mbed-client-c/source/libCoap/src/sn_coap_protocol.c: In function 'sn_coap_protocol_in
it':
C:/cygwin/home/ausbla01/mbed/mbed-client-examples/yotta_modules/mbed-client-c/source/libCoap/src/sn_coap_protocol.c:220:5: warning: implicit declarat
ion of function 'time' [-Wimplicit-function-declaration]
     srand(rand()^time(NULL));
     ^
C:/cygwin/home/ausbla01/mbed/mbed-client-examples/yotta_modules/mbed-client-c/source/libCoap/src/sn_coap_protocol.c: In function 'sn_coap_protocol_se
t_block_size':
C:/cygwin/home/ausbla01/mbed/mbed-client-examples/yotta_modules/mbed-client-c/source/libCoap/src/sn_coap_protocol.c:230:49: warning: unused parameter
 'block_size' [-Wunused-parameter]
 int8_t sn_coap_protocol_set_block_size(uint16_t block_size)
                                                 ^
C:/cygwin/home/ausbla01/mbed/mbed-client-examples/yotta_modules/mbed-client-c/source/libCoap/src/sn_coap_protocol.c: In function 'sn_coap_protocol_se
t_duplicate_buffer_size':
C:/cygwin/home/ausbla01/mbed/mbed-client-examples/yotta_modules/mbed-client-c/source/libCoap/src/sn_coap_protocol.c:252:59: warning: unused parameter
 'message_count' [-Wunused-parameter]
 int8_t sn_coap_protocol_set_duplicate_buffer_size(uint8_t message_count)
                                                           ^
[31/319] Building C object ym/mbed-client-c/source/CMakeFiles/mbed-clien...ient-examples/yotta_modules/mbed-client-c/source/libNsdl/src/sn_nsdl.c.o
C:/cygwin/home/ausbla01/mbed/mbed-client-examples/yotta_modules/mbed-client-c/source/libNsdl/src/sn_nsdl.c: In function 'validate':
C:/cygwin/home/ausbla01/mbed/mbed-client-examples/yotta_modules/mbed-client-c/source/libNsdl/src/sn_nsdl.c:1565:25: warning: comparison between signe
d and unsigned integer expressions [-Wsign-compare]
         for( int i=0; i < len; i++ ){
                         ^
[293/319] Building CXX object ym/mbed-client-mbed-os/source/CMakeFiles/m...nt-examples/yotta_modules/mbed-client-mbed-os/source/m2mtimerpimpl.cpp.o
In file included from C:/cygwin/home/ausbla01/mbed/mbed-client-examples/yotta_modules/mbed-client-mbed-os/mbed-client-mbed-os/m2mtimerpimpl.h:20:0,
                 from C:/cygwin/home/ausbla01/mbed/mbed-client-examples/yotta_modules/mbed-client-mbed-os/source/m2mtimerpimpl.cpp:16:
[... 23 lines cut ...]
C:/cygwin/home/ausbla01/mbed/mbed-client-examples/yotta_modules/sal/mbed-net-socket-abstract/socket_api.h:19:2: warning: #warning mbed-net-socket-abs
tract/socket_api.h is deprecated. Please use sal/socket_api.h instead. [-Wcpp]
 #warning mbed-net-socket-abstract/socket_api.h is deprecated.  Please use sal/socket_api.h instead.
  ^
[316/319] Building CXX object ym/mbed-client/source/CMakeFiles/mbed-clie...-client-examples/yotta_modules/mbed-client/source/m2mnsdlinterface.cpp.o
C:/cygwin/home/ausbla01/mbed/mbed-client-examples/yotta_modules/mbed-client/source/m2mnsdlinterface.cpp:339:73: warning: unused parameter 'nsdl_handl
e' [-Wunused-parameter]
 uint8_t M2MNsdlInterface::received_from_server_callback(struct nsdl_s * nsdl_handle,
                                                                         ^
C:/cygwin/home/ausbla01/mbed/mbed-client-examples/yotta_modules/mbed-client/source/m2mnsdlinterface.cpp: In member function 'virtual void M2MNsdlInte
rface::value_updated(M2MBase*)':
C:/cygwin/home/ausbla01/mbed/mbed-client-examples/yotta_modules/mbed-client/source/m2mnsdlinterface.cpp:738:15: warning: enumeration value 'ResourceI
nstance' not handled in switch [-Wswitch]
         switch(base->base_type()) {
               ^
[318/319] Building CXX object source/CMakeFiles/mbed-client-examples.dir/C_/cygwin/home/ausbla01/mbed/mbed-client-examples/source/main.cpp.o
In file included from C:/cygwin/home/ausbla01/mbed/mbed-client-examples/source/main.cpp:18:0:
C:/cygwin/home/ausbla01/mbed/mbed-client-examples/yotta_modules/mbed-drivers/mbed/test_env.h:19:2: warning: #warning mbed/test_env.h is deprecated. P
lease use mbed-drivers/test_env.h instead. [-Wcpp]
 #warning mbed/test_env.h is deprecated.  Please use mbed-drivers/test_env.h instead.
  ^
[319/319] Linking CXX executable source/mbed-client-examples

(workspace) C:\cygwin\home\ausbla01\mbed\mbed-client-examples>
```

rainierwolfcastle, 2015-11-13:
ARM Internal Ref: IOTCLT-415

teetak01, 2015-12-18:
We are working on cleaning up the compiler warnings. Thank you for reporting.

janjongboom, 2016-02-16:
All warnings are gone in this fork http://github.com/ARMmbed/mbed-client-mbedos-example/, let's see what we should do with it.

closed 2016-06-27 by anttiylitokola