ARMmbed/mbed-client-quickstart #35: Example is too complicated

BlackstoneEngineering, 2015-11-12:
This example needs to be simplified, 50 lines or less for a golden example. Its too long and too complicated to expect people to use. It needs to be wrapped up behind a simple interface library.

rainierwolfcastle, 2015-11-13:
ARM Internal Ref: IOTCLT-413

teetak01, 2015-12-18:
Current example demonstrates several functionalities of the client and as such cannot be implemented in much shorter format. Should there be another simpler example to demonstrate minimal functionality?

BlackstoneEngineering, 2015-12-18:
@teetak01 the Apps team is moving forward with more concise examples, Its on our list for january, no worries. Something that is clearer, like this : https://github.com/BlackstoneEngineering/mbed-client-examples

note the simplification of the code and the heavy commenting. It was used in the meetup documented here : https://www.youtube.com/watch?v=nWz9Toe8nJE

janjongboom, 2016-02-16:
Clearer example is now here: http://github.com/ARMmbed/mbed-client-mbedos-example/ although we need to think of upgrade path. Maybe merge that back in this repo?

teetak01, 2016-02-16:
In my opinion the mbed-client-examples is a bit confusingly named as it does not work on linux. Maybe the new name is better?

There are now two different naming schemes, we have:

https://github.com/ARMmbed/mbed-client-example-6lowpan

and

https://github.com/ARMmbed/mbed-client-linux-example

Maybe formalize the naming along similar lines, for. ex. mbed-client-example-XXX

anttiylitokola, 2016-04-04:
Clearer example is taken into use. This can be closed now.

closed 2016-04-04 by anttiylitokola