ARMmbed/mbed-client-quickstart #37: Compile-Linking failed in debug mode

cyliangtw, 2015-11-30:
I set target as frdm-k64f-armcc and build by "yotta build", it could connect/register connector successfully. For debug,  to build by ""yotta build --debug-build", I got error as "Undefined symbol platform_enter_critical". So, I implemented the undefined symbol as:
```
 // disable global interrupts
 void platform_enter_critical(void)
 { __disable_irq();}
 
 //enable global interrupts
 void platform_exit_critical(void)
 { __enable_irq();}
```
After build successfully, to run this sample by " yotta debug mbed-client-examples".
It will encounter HardFault after got IP address.

anttiylitokola, 2015-12-01:
Thanks for your report. It seems that there are couple of things missing in tracing that we must enable to get debugging work. We will let you know once that is done.

anttiylitokola, 2016-01-15:
Hi, this has been now fixed and you should be able to compile using debug option.

closed 2016-01-15 by anttiylitokola