ARMmbed/mbed-client-quickstart #40: Can't get the IP address

sunsmilearm, 2016-01-12:
I compiled this project and downloaded it into k64f. The output in terminal is only "IP address". It seems the K64F can't get its IP using DHCP. The internet cable is OK, my laptop or raspberry Pi can get IP address.
Thanks.

yogpan01, 2016-01-12:
Hi , can you please paste your "yotta ls" . We would like to see which version of sockets library you are using.

sunsmilearm, 2016-01-14:
The yotta ls result:
C:\my\dev\mbed-client-examples-k64f>yotta ls
mbed-client-examples 1.1.0
\_ mbed-client 1.2.16
  |_ mbed-client-c 1.1.2 yotta_modules\mbed-client-c
  | \_ nanostack-libservice 3.0.10 yotta_modules\nanostack-libservice
  |_ mbed-client-mbed-os 1.1.1 yotta_modules\mbed-client-mbed-os
  | |_ mbed-drivers 0.11.6 yotta_modules\mbed-drivers
  | | |_ mbed-hal 1.2.0 yotta_modules\mbed-hal
  | | | \_ mbed-hal-freescale 1.0.0 yotta_modules\mbed-hal-freescale
  | | |   \_ mbed-hal-ksdk-mcu 1.0.6 yotta_modules\mbed-hal-ksdk-mcu
  | | |     |_ uvisor-lib 1.0.11 yotta_modules\uvisor-lib
  | | |     \_ mbed-hal-k64f 1.0.1 yotta_modules\mbed-hal-k64f
  | | |       \_ mbed-hal-frdm-k64f 1.0.0 yotta_modules\mbed-hal-frdm-k64f
  | | |_ cmsis-core 1.1.0 yotta_modules\cmsis-core
  | | | \_ cmsis-core-freescale 1.0.0 yotta_modules\cmsis-core-freescale
  | | |   \_ cmsis-core-k64f 1.0.0 yotta_modules\cmsis-core-k64f
  | | |_ ualloc 1.0.3 yotta_modules\ualloc
  | | | \_ dlmalloc 1.0.0 yotta_modules\dlmalloc
  | | |_ minar 1.0.1 yotta_modules\minar
  | | | \_ minar-platform 1.0.0 yotta_modules\minar-platform
  | | |   \_ minar-platform-mbed 1.0.0 yotta_modules\minar-platform-mbed
  | | |_ core-util 1.1.0 yotta_modules\core-util
  | | \_ compiler-polyfill 1.2.1 yotta_modules\compiler-polyfill
  | \_ sockets 1.1.0 yotta_modules\sockets
  |   \_ sal 1.1.1 yotta_modules\sal
  |     \_ sal-stack-lwip 1.1.0 yotta_modules\sal-stack-lwip
  |       |_ sal-driver-lwip-k64f-eth 1.0.2 yotta_modules\sal-driver-lwip-k64f-e
th
  |       \_ sal-iface-eth 1.0.1 yotta_modules\sal-iface-eth
  \_ mbed-client-mbedtls 1.0.11 yotta_modules\mbed-client-mbedtls
    \_ mbedtls 2.2.0 yotta_modules\mbedtls

anttiylitokola, 2016-01-15:
Hi, can you run yotta up to fetch latest release and try again?

sunsmilearm, 2016-01-18:
Hi, thanks for your reply. I run yotta up and some modules are updated. But K64F still can't get the IP address. Here is the result of yotta ls:

mbed-client-examples 1.1.0
\_ mbed-client 1.2.18
  |_ mbed-client-c 2.2.0 yotta_modules\mbed-client-c
  | \_ nanostack-libservice 3.0.10 yotta_modules\nanostack-libservice
  |_ mbed-client-mbed-os 1.1.1 yotta_modules\mbed-client-mbed-os
  | |_ mbed-drivers 0.11.8 yotta_modules\mbed-drivers
  | | |_ mbed-hal 1.2.1 yotta_modules\mbed-hal
  | | | \_ mbed-hal-freescale 1.0.0 yotta_modules\mbed-hal-freescale
  | | |   \_ mbed-hal-ksdk-mcu 1.0.7 yotta_modules\mbed-hal-ksdk-mcu
  | | |     |_ uvisor-lib 1.0.12 yotta_modules\uvisor-lib
  | | |     \_ mbed-hal-k64f 1.0.1 yotta_modules\mbed-hal-k64f
  | | |       \_ mbed-hal-frdm-k64f 1.0.0 yotta_modules\mbed-hal-frdm-k64f
  | | |_ cmsis-core 1.1.1 yotta_modules\cmsis-core
  | | | \_ cmsis-core-freescale 1.0.0 yotta_modules\cmsis-core-freescale
  | | |   \_ cmsis-core-k64f 1.0.0 yotta_modules\cmsis-core-k64f
  | | |_ ualloc 1.0.3 yotta_modules\ualloc
  | | | \_ dlmalloc 1.0.0 yotta_modules\dlmalloc
  | | |_ minar 1.0.1 yotta_modules\minar
  | | | \_ minar-platform 1.0.0 yotta_modules\minar-platform
  | | |   \_ minar-platform-mbed 1.0.0 yotta_modules\minar-platform-mbed
  | | |_ core-util 1.1.1 yotta_modules\core-util
  | | \_ compiler-polyfill 1.2.1 yotta_modules\compiler-polyfill
  | \_ sockets 1.1.0 yotta_modules\sockets
  |   \_ sal 1.1.1 yotta_modules\sal
  |     \_ sal-stack-lwip 1.1.0 yotta_modules\sal-stack-lwip
  |       |_ sal-driver-lwip-k64f-eth 1.0.2 yotta_modules\sal-driver-lwip-k64f-e
th
  |       \_ sal-iface-eth 1.0.1 yotta_modules\sal-iface-eth
  \_ mbed-client-mbedtls 1.0.11 yotta_modules\mbed-client-mbedtls
    \_ mbedtls 2.2.0 yotta_modules\mbedtls

ciscoski, 2016-01-19:
Same problem here:
When I connect the Serial Terinal I get :
```
IP address 
```
```
ciscoski@Ciscoski-Air-2:~/Sviluppo/mbed/mbed-client-examples$ yotta ls
mbed-client-examples 1.1.0
┣━ mbed-client 1.2.18
┃ ┣━ mbed-client-c 2.2.0 yotta_modules/mbed-client-c
┃ ┃ ┗━ nanostack-libservice 3.0.10 yotta_modules/nanostack-libservice
┃ ┣━ mbed-client-mbed-os 1.1.1 yotta_modules/mbed-client-mbed-os
┃ ┃ ┗━ sockets 1.1.0 yotta_modules/sockets
┃ ┃   ┣━ sal 1.1.1 yotta_modules/sal
┃ ┃   ┃ ┣━ cmsis-core 1.1.1 yotta_modules/cmsis-core
┃ ┃   ┃ ┃ ┗━ cmsis-core-freescale 1.0.0 yotta_modules/cmsis-core-freescale
┃ ┃   ┃ ┃   ┗━ cmsis-core-k64f 1.0.0 yotta_modules/cmsis-core-k64f
┃ ┃   ┃ ┗━ sal-stack-lwip 1.1.1 yotta_modules/sal-stack-lwip
┃ ┃   ┃   ┣━ sal-driver-lwip-k64f-eth 1.0.2 yotta_modules/sal-driver-lwip-k64f-eth
┃ ┃   ┃   ┗━ sal-iface-eth 1.0.1 yotta_modules/sal-iface-eth
┃ ┃   ┣━ core-util 1.1.1 yotta_modules/core-util
┃ ┃   ┃ ┗━ ualloc 1.0.3 yotta_modules/ualloc
┃ ┃   ┃   ┗━ dlmalloc 1.0.0 yotta_modules/dlmalloc
┃ ┃   ┗━ minar 1.0.1 yotta_modules/minar
┃ ┃     ┣━ compiler-polyfill 1.2.1 yotta_modules/compiler-polyfill
┃ ┃     ┗━ minar-platform 1.0.0 yotta_modules/minar-platform
┃ ┃       ┗━ minar-platform-mbed 1.0.0 yotta_modules/minar-platform-mbed
┃ ┃         ┗━ mbed-hal 1.2.1 yotta_modules/mbed-hal
┃ ┃           ┗━ mbed-hal-freescale 1.0.0 yotta_modules/mbed-hal-freescale
┃ ┃             ┗━ mbed-hal-ksdk-mcu 1.0.7 yotta_modules/mbed-hal-ksdk-mcu
┃ ┃               ┣━ uvisor-lib 1.0.12 yotta_modules/uvisor-lib
┃ ┃               ┗━ mbed-hal-k64f 1.0.1 yotta_modules/mbed-hal-k64f
┃ ┃                 ┗━ mbed-hal-frdm-k64f 1.0.0 yotta_modules/mbed-hal-frdm-k64f
┃ ┗━ mbed-client-mbedtls 1.0.11 yotta_modules/mbed-client-mbedtls
┃   ┗━ mbedtls 2.2.0 yotta_modules/mbedtls
```
I've tried the HTTPServer_echoback example and it works without any issue.

anttiylitokola, 2016-01-20:
Just to be sure, is this the application which was able to get IP address successfully?
https://developer.mbed.org/users/hsgw/code/HTTPServer_echoback/file/eae1575da9ca/main.cpp

Quite interesting since we are using the same code for initializing ethernet connection. 
If you can do a small modification to code(main.cpp) so we can see is the connect() causing the problem or something else.
....
    int err = eth.init();
    if (err != 0) {
        output.printf("Initialization failed: %d\r\n", err);
    }
    err = eth.connect();
    if (err != 0 ) {
        output.printf("Connection failed: %d\r\n", err);
        }
    err = lwipv4_socket_init();
    if (err != 0) {
        output.printf("Error on lwipv4_socket_init: %d\r\n", err);
    }

output.printf("MAC address %s\r\n", eth.getMACAddress());
output.printf("IP address %s\r\n", eth.getIPAddress());
output.printf("Device name %s\r\n", MBED_ENDPOINT_NAME);
....

One thing worth to try is to flash latest firmware if you haven't done that yet. Instructions how to do that can be found from following link:
https://developer.mbed.org/handbook/Firmware-FRDM-K64F

Thanks.

ciscoski, 2016-01-20:
Yes that
```
Connection failed: -1
MAC address 15:32:70:0b:00:1d
IP address 
Device name 485c7f92-b117-488c-8f62-e773137aeda6
```

I've already updated the firmware.
```
Version: 0226
Build:   Aug 24 2015 17:06:30
Git Commit SHA: 27a236b9fe39c674a703c5c89655fbd26b8e27e1
Git Local mods: Yes
```

stevenkroon, 2016-01-31:
I've got the exact same issue, upgraded firmware to 0226 and also getting "Failed to form a connection!"

anttiylitokola, 2016-02-01:
Hi, we have identified the problem and there are now two pull request's waiting to be merged. These should fix the IP address problem.
Links to PR's:
https://github.com/ARMmbed/mbed-hal-k64f/pull/24
https://github.com/ARMmbed/mbed-hal-frdm-k64f/pull/15

stevenkroon, 2016-02-03:
I've applied the patches and it works! I'm getting an IP address, currently still getting a NetwokError but that might not be related to the issue here.

```
Connected!
IP address is 192.168.178.24
Device name lwm2m-endpoint

[ERROR:] M2MInterface::NetworkError
```

bremoran, 2016-02-10:
The PRs listed above have now been merged.

closed 2016-02-10 by anttiylitokola