ARMmbed/mbed-client-quickstart #44: Example for value_updated()

markus-becker-tridonic-com, 2016-01-26:
The MbedClient is getting informed with value_updated() about a change by a PUT.

How does one use the information provided in value_updated to inform the respective object/object_instance/resource of the change? An example would be appreciated.

mlnx, 2016-01-26:
@yogpan01 could you please comment on this?

yogpan01, 2016-01-26:
Hi, In the callback function of value_updated() you get the M2MBase* object as well as the object type (whether its Object or ObjectInstance or Resource or ResourceInstance) .You can then typecast into the respective Object Type and extract out the required value. We will soon update example code as to how you can get the value out.

markus-becker-tridonic-com, 2016-01-26:
How do you know to which respective Object Type to cast to? Please let me know about example code.

yogpan01, 2016-01-27:
Hi, can you try like this ?

```
void value_updated(M2MBase *base, M2MBase::BaseType type) {
        uint16_t instance_id = base->instance_id();

        M2MObjectInstance *object_instance = <your_object>->object_instance(instance_id);
        M2MResource *resource = object_instance->resource(<resource_id>);
}
```

ivan-adzic-tridonic-com, 2016-01-27:
Hi, We have tried it and instance_id is always 0, even if the resource value is changed for the object which has instance_id 1.

yogpan01, 2016-01-27:
The base type can tell you which type of object was changed. But I guess you need complete information of which Object ID, ObjectInstance ID and Resource ID has changed. We can extend this API so that it can return detailed information so it will be easy to map the object.

BlackstoneEngineering, 2016-02-23:
You shouldn't use the value_updated() fn to trigger data execution. PUT messages are not meant to trigger anything, they are just a means of keeping data in synch between mbed connector and mbed client. To trigger an execution you should POST to a resource. 

IE send the firmware to the device by PUTing to the /firmware resource, and triggering a firmware update by POSTing to the /firmware or /firmware-update resource. 

The current mbed-client stack is not meant to trigger actions on PUT's

markus-becker-tridonic-com, 2016-03-14:
That would mean more or less every PUT would be a POST. When you are PUTting a configuration value, the application should be informed. Other stacks (e.g. Contiki) do that (additionally also for GET).

BlackstoneEngineering, 2016-03-14:
No, its means you should have PUT /POST enabled for the resource, and use the POST execute callback to inform your state. So yes, under the current structure you have to PUT then POST to trigger. Or you can just post with data. 

This is an architectual issue and a discussion for the mbed-client team.

yogpan01, 2016-03-14:
@markus-becker-tridonic-com As mentioned earlier, we propose to add helper API to identify the resources who's value has changed. Check this PR https://github.com/ARMmbed/mbed-client/pull/170
After this , you can get details about the object ID and object instance to which the changed resource belongs to.

anttiylitokola, 2016-03-21:
Hi, PR https://github.com/ARMmbed/mbed-client/pull/170 is now published. Documentation how it can be used can be found from https://github.com/ARMmbed/mbed-client/blob/anttiylitokola-patch-1/docs/dev_man_serv_enable.md#the-write-operation

closed 2016-03-21 by anttiylitokola