ARMmbed/mbed-client-quickstart #45: Example of execute_callback for object with more than 1 instance.

ivan-adzic-tridonic-com, 2016-01-27:
The M2MObject is getting informed with execute_callback() about a change by a POST from handle_post_request.

Our object have several instances. Lets say 4. 
How we can get information about for which instance_id we should execute callback?

If we call within the registered execute callback this->instance_id(); we will always get same id which is 0.

An example would be appreciated.

yogpan01, 2016-01-27:
How about if you try like this ?

```
void value_updated(M2MBase *base, M2MBase::BaseType type) {
        uint16_t instance_id = base->instance_id();

        M2MObjectInstance *object_instance = <your_object>->object_instance(instance_id);
        M2MResource *resource = object_instance->resource(<resource_id>);
}
````

ivan-adzic-tridonic-com, 2016-01-27:
Hi, in our execute callbacks we only have *arguments (void M2MBase::execute(void *arguments)) 
as input parameters, not (M2MBase *base, M2MBase::BaseType type).

This is probably an answer for previous issue " Example for value_updated() #44 "

yogpan01, 2016-01-27:
OK, got it. You can map the passing function with your resource or then we can extend this API so that it can contain more information about resource.

anttiylitokola, 2016-03-21:
Hi, PR ARMmbed/mbed-client#170 is now published. Documentation how it can be used can be found from https://github.com/ARMmbed/mbed-client/blob/anttiylitokola-patch-1/docs/dev_man_serv_enable.md#the-execute-operation

closed 2016-03-21 by anttiylitokola