ARMmbed/mbed-client-quickstart #49: Error building

danielamorais, 2016-02-21:
Hi

This is the first time I try to run this sample so already I apologize if it's something stupid. 
I have installed _yotta 0.13.1_, _Cmake 3.4.1_, _clang 3.7_ and _GCC ARM Embedded 5.0_ but I get the error below. 
I need to run before 

> yotta target x86 -linux -native

 or does not make sense?

```
[danielamorais@localhost mbed-client-examples-master]$ yotta target frdm-k64f-gcc
[danielamorais@localhost mbed-client-examples-master]$ yotta build
info: generate for target: frdm-k64f-gcc 2.0.0 at /home/danielamorais/Downloads/mbed-client-examples-master/yotta_targets/frdm-k64f-gcc
GCC version is: 5.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/danielamorais/Downloads/mbed-client-examples-master/build/frdm-k64f-gcc
[8/323] Building C object ym/nanostack-libservice/source/CMakeFiles/nanostack-libs...-examples-master/yotta_modules/nanostack-libservice/source/libip6string/stoip6.c.o
/home/danielamorais/Downloads/mbed-client-examples-master/yotta_modules/nanostack-libservice/source/libip6string/stoip6.c: In function 'stoip6':
/home/danielamorais/Downloads/mbed-client-examples-master/yotta_modules/nanostack-libservice/source/libip6string/stoip6.c:44:42: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (field_no = 0, p = ip6addr; (len > (p - ip6addr)) && *p && field_no < 8; p = q + 1) {
                                          ^
/home/danielamorais/Downloads/mbed-client-examples-master/yotta_modules/nanostack-libservice/source/libip6string/stoip6.c:54:18: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if ((len > (q - ip6addr)) && *q && (q[0] == ':') && (q[1] == ':')) {
                  ^
[15/323] Linking C static library ym/nanostack-libservice/source/nanostack-libservice.a
FAILED: : && CMAKE_AR-NOTFOUND -cr  ym/nanostack-libservice/source/nanostack-libservice.a ym/nanostack-libservice/source/CMakeFiles/nanostack-libservice.dir/home/danielamorais/Downloads/mbed-client-examples-master/yotta_modules/nanostack-libservice/source/libip6string/ip6tos.c.o ym/nanostack-libservice/source/CMakeFiles/nanostack-libservice.dir/home/danielamorais/Downloads/mbed-client-examples-master/yotta_modules/nanostack-libservice/source/libList/ns_list.c.o ym/nanostack-libservice/source/CMakeFiles/nanostack-libservice.dir/home/danielamorais/Downloads/mbed-client-examples-master/yotta_modules/nanostack-libservice/source/libTrace/ns_trace.c.o ym/nanostack-libservice/source/CMakeFiles/nanostack-libservice.dir/home/danielamorais/Downloads/mbed-client-examples-master/yotta_modules/nanostack-libservice/source/libBits/common_functions.c.o ym/nanostack-libservice/source/CMakeFiles/nanostack-libservice.dir/home/danielamorais/Downloads/mbed-client-examples-master/yotta_modules/nanostack-libservice/source/libip6string/stoip6.c.o ym/nanostack-libservice/source/CMakeFiles/nanostack-libservice.dir/home/danielamorais/Downloads/mbed-client-examples-master/yotta_modules/nanostack-libservice/source/IPv6_fcf_lib/ip_fsc.c.o ym/nanostack-libservice/source/CMakeFiles/nanostack-libservice.dir/home/danielamorais/Downloads/mbed-client-examples-master/yotta_modules/nanostack-libservice/source/nsdynmemLIB/nsdynmemLIB.c.o && :
/bin/sh: CMAKE_AR-NOTFOUND: command not found
[15/323] Building C object ym/dlmalloc/source/CMakeFiles/dlmalloc.dir/home/daniela...s/Downloads/mbed-client-examples-master/yotta_modules/dlmalloc/source/dlmalloc.c.o
ninja: build stopped: subcommand failed.
error: command ['ninja-build'] failed

```

Thanks

yogpan01, 2016-02-21:
Hi,
The example is for FRDM K64F board. So, you need to set your yotta target as 
`yotta target frdm-k64f-gcc`

danielamorais, 2016-02-21:
Any ideia why I get `CMAKE_AR-NOTFOUND: command not found`?

danielamorais, 2016-02-21:
Fixed

```
[danielamorais@localhost bin]$ export PATH=$PATH:/home/danielamorais/Downloads/gcc-arm-none-eabi-5_2-2015q4/bin/arm-none-eabi-ar
[danielamorais@localhost bin]$ cd /usr/bin/
[danielamorais@localhost bin]$ sudo ln -s /home/danielamorais/Downloads/gcc-arm-none-eabi-5_2-2015q4/bin/arm-none-eabi-ar arm-none-eabi-ar
```

Thanks

closed 2016-02-22 by yogpan01