ARMmbed/mbed-client-quickstart #50: Can No Longer LWM2M Register

SRomeroamI, 2016-03-02:
Hi,

I have been using this example quite a bit with no problem at all with UDP and certificate mode using a Freescale FRDMK64 board.    

I ran into some problems successfully registering (LWM2M registration) on a different project using the same Freescale board after doing a yotta clean yotta build.   

After some troubleshooting, I decided to do a yotta clean yotta build on this project - just to check everything is OK.   For some reason, I can no longer reach the LWM2M Registration state on this project either.

I deleted the entire project and re-cloned it, re-installed the latest yotta, and I have checked that my security.h is a direct copy from mbed-connector.   Yet, the results are the same.

I am not sure what else to do.  Has anyone else reported a similar problem?  

Regards,

SR

yogpan01, 2016-03-02:
Hi, Thanks for reporting. We will be looking into this. Can you please post your dependency list that you can get through "yotta ls" ?

teetak01, 2016-03-02:
Hi, could you do "yotta update" and rebuild and re-flash? I think we managed to fix the issue (LWIP-driver issue). Sorry for the inconvenience.

SRomeroamI, 2016-03-02:
Hi, I followed your instructions and both projects work again!!!   Thanks very much for your quick response.

SR

closed 2016-03-02 by SRomeroamI