ARMmbed/mbed-client-quickstart #63: RAM overflow for frdm-k64f-gcc target

maxpeng, 2016-04-27:
I could not build this example with arm gcc (5_3_2016q1 release) on Windows, it failed with following error:
```
.../5_3_2016q1/bin/../lib/gcc/arm-none-eabi/5.3.1/../../../../arm-none-eabi/bin/ld.exe: source/mbed-client-quickstart section `.bss' will not fit in region `RAM'

.../5_3_2016q1/bin/../lib/gcc/arm-none-eabi/5.3.1/../../../../arm-none-eabi/bin/ld.exe: region `RAM' overflowed by 8966 bytes

collect2.exe: error: ld returned 1 exit status

ninja: build stopped: subcommand failed.
error: command ['ninja'] failed
```

teetak01, 2016-04-27:
Hi @maxpeng . Thank you for reporting this. As an immediate fix to this, you can change one line in target module:

`yotta_targets/frdm-k64f-gcc/target.json`

There change the `"kinetis-k64-gcc": "^2.0.0"` line to this=>    `"kinetis-k64-gcc": "2.0.2"`

After that you should be able to build the example. Apologies for this. The issue will be resolved promptly.

maxpeng, 2016-04-28:
With the change as you described, I am able to build the example.

But I don't know why the change had been reverted for **kiletis-k64f-gcc v2.1.0**, please see [Merge pull request #5 from ARMmbed/fix_non_contingent_ram ](https://github.com/ARMmbed/target-kinetis-k64-gcc/commit/e1eb5bf482e0e74b21708307de18e5bb084a6154).

teetak01, 2016-04-28:
Fix is now in yotta. Yotta update should be enough. If not `rm -rf yotta_targets` and `yt clean`.

teetak01, 2016-04-28:
@maxpeng Let me know if you still have any issues.

maxpeng, 2016-04-28:
Hi Teetake01, no problem now, thanks!

closed 2016-04-28 by maxpeng

maxpeng, 2016-04-28:
Sorry, I have one question about the fix.

Target **frdm-k64f-gcc** inherits **kinetics-64f-gcc**, I thought *frdm-k64f-gcc/ld/K64FN1M0xxx12.ld* would override *kinetics-64f-gcc/ld/K64FN1M0xxx12.ld*, why does the fix update *kinetics-64f-gcc/ld/K64FN1M0xxx12.ld* instead of *frdm-k64f-gcc/ld/K64FN1M0xxx12.ld*?

When will the linker use *frdm-k64f-gcc/ld/K64FN1M0xxx12.ld*?

teetak01, 2016-05-03:
Hi @maxpeng that is a good question. I could not figure out if it is ever used. In fact, I opened a ticket on that for the target to check if there is something missing.

https://github.com/ARMmbed/target-frdm-k64f-gcc/issues/26

maxpeng, 2016-05-04:
Hi @teetak01, thanks!