ARMmbed/mbed-client-quickstart #67: Secure Connection Failed Error

emilijatasic, 2016-06-10:
Hello. 
I have been trying to run this example, following the instructions on readme (c/p security credentials, etc.) , but I keep getting this error. 
 I am using a university network. 
[image]

Thanks!

yogpan01, 2016-06-11:
Hi,
It seems that there is issue with DTLS Handshake on UDP , did you try using TCP mode ? You can change socket mode by following these instructions https://github.com/ARMmbed/mbed-client-quickstart/blob/master/README.md#Setting-socket-type

emilijatasic, 2016-06-11:
I tried that too, but with same result

yogpan01, 2016-06-11:
You can also try to build the log enabled build . 
Try building with "yotta build -d" , this should enable detailed logs , which might reveal some more information about this error.

emilijatasic, 2016-06-11:
Ok, I will try that, thanks! 
I will update this topic on monday.

emilijatasic, 2016-06-13:
So I just tried to build with yotta build -d, and got this error on terminal(in bunch of other information):

[ERR ][mClt]: M2MConnectionSecurityPimpl::timer_expired - handshake timeout

What does that mean , or - what should I do to solve this issue?

teetak01, 2016-06-13:
Hi, do you know if the university network has the port 5684 open? It might be that connector is unable to respond.

emilijatasic, 2016-06-14:
The port 5684 hasn't been open, but now it is and everything works now. Thanks

closed 2016-06-14 by yogpan01