ARMmbed/mbed-client-quickstart #69: [ERROR:] M2MInterface::SecureConnectionFailed

jsplyy, 2016-07-04:
I run the example and do nothing revisions on the frdm-k64
it keep waitting on the connectting...
Timeout and print error:[ERROR:] M2MInterface::SecureConnectionFailed
So i try to replace the domin address with ip adress directly. But it is still error.

Anyone knows the reason?

    In app_start()
	IP address 10.193.101.111
	Device name 277683b2-0468-4690-a7b5-582129057200

	SOCKET_MODE : TCP
	Connecting to coap://169.45.82.18:5684

	[ERROR:] M2MInterface::SecureConnectionFailed

yogpan01, 2016-07-04:
It seems that your TLS handshake is failing, do you have any firewall which might prevent data handling on port 5684 ?

jsplyy, 2016-07-04:
@yogpan01 yes,I go back home and try it again. It works well using ip or domain.
That erro is caused by company firewall.
Thanks for your response.

closed 2016-07-06 by yogpan01

jsplyy, 2016-07-07:
Hi, I talk with safe group and they tell me that my internet access is always allowed.
They let me ask whether your server Firewall block?