ARMmbed/mbed-client-quickstart #70: [ERROR:] M2MInterface::SecureConnectionFailed

jsplyy, 2016-07-07:
Hi, I talk with safe group and they tell me that my internet access is always allowed.
They let me ask whether your server Firewall block?

jsplyy, 2016-07-07:
Here IP is 123.151.195.1

yogpan01, 2016-07-07:
Hi, Connector is open for accepting all incoming connection requests. It sometime happens certain ports might be blocked by ISPs. What you can ensure that your ISP can accept and send network data on all possible ports. 
You can test from using Linux based mbed Client example application , if your network connection are working towards Connector.

jsplyy, 2016-07-07:
Ok, I will try later. If the ISPs block, how to deal with it?

jsplyy, 2016-07-08:
hi , I check all the issue and contact with relative people who all tell me net is ok.
yesterday night, it suddenly connects to the connector server.
But today is still error.

anttiylitokola, 2016-08-02:
Hi, are you still experiencing connection related problems? If not, think we can close this ticket.

closed 2016-08-11 by anttiylitokola