ARMmbed/mbed-client-quickstart #71: yt target frdm-k64f-armcc build fail

jsplyy, 2016-07-07:
I want to debug the mbed os in keil. So I build it in armcc target.
But it fail. 

[image]

jsplyy, 2016-07-07:
@ciarmcom  what's mean?

yogpan01, 2016-07-07:
Hi, 
Do you have ARM Compiler installed ? You need to install it .
Here is the link 
https://developer.arm.com/products/software-development-tools/compilers/arm-compiler-5/downloads

Remember this is a commercial compiler so you would need proper license as well. Check the link, it has description about it.

jsplyy, 2016-07-28:
@yogpan01  Hi.
Now, I have installed keil-mdk and could use armcc compile.
But when I debug,the mdk does not appear as descriped in docs.
[image]

jsplyy, 2016-07-28:
I want use an IDE to develop mbed os.
which is supported and how to support it ?

jsplyy, 2016-07-28:
When will Online compiler be revamped?

teetak01, 2016-08-22:
@jsplyy Hi, the Yotta-based example does not directly support any IDE's. 

Also, have you taken a look to our newer example program using the latest mbed-os? https://github.com/armmbed/mbed-os-example-client

Its build-environment also supports IDE integration: https://github.com/armmbed/mbed-cli#exporting-to-desktop-ides

Do you still have any issues related to this repository or can this be closed?

closed 2016-09-02 by yogpan01