ARMmbed/mbed-client-quickstart #73: multi k64 boards connect to connector server error.[ERROR:] M2MInterface::InvalidParameters

jsplyy, 2016-07-13:
I try to connect multi boards to connector server.
If I do not modify the MBED_ENDPOINT_NAME "277683b2-0468-4690-a7b5-582129057200",it will show one device in connector server.
But if I modify the MBED_ENDPOINT_NAME "277683b2-0468-4690-a7b6-582129057200",it will show error[ERROR:] M2MInterface::InvalidParameters

What's the relationship between mbed endpoint name and mbed domain?
[image]

yogpan01, 2016-07-14:
Hi,
You cannot modify endpoint name yourself, those endpoint names are embedded into certificates , so if those are mismatching ,server will not accept those endpoints as valid ones, hence the errors.
You need unique security.h for each devices.

jsplyy, 2016-07-14:
hmm...
I want to connect multi boards. How to do?

anttiylitokola, 2016-08-02:
Hi, you can create a new security.h for every board you need. Just click the "GET MY SECURITY CREDENTIALS" button in Connector website. It will create a new security.h containing unique endpoint name. So every board must have unique security.h.

closed 2016-08-11 by anttiylitokola